irllabs/eagle #10: GND 0.9V higher than GND1 for breakout circuit

Preetisar, 2017-12-23:
For a functioning breakout circuit, there is a potential difference of about 0.9 V between GND and GND1, ie, they are not shorted. Can this please be explained.

jakezimmer, 2018-01-05:
Hi Preeti, this is fine. The mosfets incorperated into the chip that we are using for voltage and current monitoring of the battery are doing something called low side switching (more info here: https://learn.sparkfun.com/tutorials/transistors/applications-i-switches). 

This means that when something is wrong with the battery, the mosfet disconnects the negative of the battery (GND1) from the real ground (GND). This explains the 3.9v error because the battery was probably charged to ~3.9v. 

For the 0.9v difference, sometimes mosfets will have rather larger (but somewhat constant) voltage drops across them characteristic of many other semiconductors (diodes, LEDs). This is expected behavior in this chip and so since we are stepping it up anyway and our circuit is low power, doesn't really matter too much. Were this a motor driver or something high power, we might want to look into something else.

Preetisar, 2018-01-06:
GND is the ground of the output terminal (which I suppose comes form the battery protector circuit) whereas GND1 is the ground from the battery. I wrote that GND is 3.9V higher than GND1, which is confusing to me because

closed 2018-01-06 by Preetisar