irllabs/eagle #14: Size of whole for the side pin of 50K dial POT in Piezo board too small

Preetisar, 2018-02-02:
The size of the side pin of the 50K dial POT is 1.02 mm which is too big for the hole in the current board.
The whole in the current board is approximately 0.41 mm

Preetisar, 2018-02-02:
The size of the pins on the 50 K dial knob is about 65 mm wide which is also too big for the current hole size on the PIEZO board (0.40 mm). But it can still be forcibly pushed in. We might want to change the whole size for all the 5 pins. 3 pins at 65 mm and 2 pins at 1 mm.

nkakarla1, 2018-02-10:
The earlier pot had the radius of 0.5 mm for side pins. New one added is changed to radius of1 mm, the closest available footprint.

Also, the 3 smaller pins are now changed to radius of 0.75 mm.

nkakarla1, 2018-02-16:
footprint of the pot updated according to the datasheet of the ordered part.

closed 2018-02-16 by Preetisar