irllabs/eagle #17: Difference in connection between Buffer Stage 1 and Op-Amp amplifier for the old and new piezo preamp boards

Preetisar, 2018-02-07:
In the old green board (piezo-preamp-tickler_V3-single-supply-OPA1612), pin 3 of the BAT54S is not connected to anything and pin 2 of the BAT54S is connected to two resistors R1, R2 and the input pin 3 of the opamp.

Whereas in the latest piezo board (piezo-preamp-tickler_V3-5_single-supply-OPA1642), pin 3 of the BAT54S is connected to two resistors R1, R2 and pin 2 of the BAT54S is connected to the 100uH inductor  L1, the input pin 3 of the opamp, capacitor C4 and resistor R3.

nkakarla1, 2018-02-10:
The board is in adherence with the new schematic. To be checked whether the changes in schematic were intentional.

Preetisar, 2018-02-10:
What do you mean by new schematic? Please mention version numbers. The latest schematic (piezo-preamp-tickler_V3-5_single-supply-OPA1642) is faulty when it comes to how the pins of the BAT54S are connected and we want to go back to how the BAT54S was connected in the older version (piezo-preamp-tickler_V3-single-supply-OPA1612) with some edits.

nkakarla1, 2018-02-13:
Connection updated: Pin 3 disconnected from all nets, pin 2 connected to R1, R2 and pin 3 of the op-amp IC1A

Preetisar, 2018-02-13:
Pin 3 of the opamp is not connected. . . Resistor R2 should be connected at the voltage divider circuit between R3 and R4. Pin 2 of the BAT54S should not be connected to that point.

nkakarla1, 2018-02-13:
The required updates made:

R2 connected between R3 and R4, pin 2 connection to that point removed.

closed 2018-02-14 by Preetisar