irllabs/eagle #39: Change Voltage regulation part of mixer board to same as piezo

Preetisar, 2018-03-28:
Name of file:
Mixer-Analog-OPA4134-Rev-K-multi under folder Mixer-Analog.

nkakarla1, 2018-04-06:
Voltage regulation part changed to same as piezo.

Preetisar, 2018-04-06:
In what version? I do not see any change in Rev L and but in Rev K. Shouldn't L come after K? Also the charging part is part of the voltage regulation (I know it is not obvious). So we need to change the charging circuit too.

nkakarla1, 2018-04-06:
Naively, yes, L must be after K and I too checked that board first.However, it has same inconsistency problem between schematic and board. Due to that, and also since it was mentioned as revision K here, I made changes to the K revision.

nkakarla1, 2018-04-06:
Yes, by voltage regulation I assumed it is the circuit which converts from one voltage level to another. Will update the charging part too and upload it.

nkakarla1, 2018-04-16:
Power circuitry updated in version M.

closed 2018-04-16 by Preetisar