irllabs/eagle #44: Change the holes of the potentiometer

Preetisar, 2018-04-08:
Change the hole diameters of the 100 K pots similar to the 20K pots in the MEMS board.
Also 50K pot is almost good but the wide distance between the two large support pins seemed a little small to me. Just to be sure please verify the distance between the two large holes for the support pins with the 
 50K pot in the piezo board (the piezo 50k is not dual gang but sizes should be the same).

nkakarla1, 2018-04-16:
pot diameter changed for 100K ones. 50K ones are fine. i.e., the distance between the support pins is same as piezo pot.

Preetisar, 2018-04-16:
Sorry but I don't see the change.

nkakarla1, 2018-04-21:
The pot footprint is changed in the source library (CMU artfab). Hence, even the older versions got updated to the new device.

Preetisar, 2018-04-23:
The diameter of the hole of the side pins of the 100K pot is not the same as the 20K pot in the MEMS board.

Preetisar, 2018-05-24:
Please make the diameter of the side pins of the 100K pots to 2.50 mm.